husarion/rosbot_ros #1: Couldn't see the robot model in Rviz

JaneCX, 2019-02-07:
it showed the urdf parsed ok, while the followings all showed "No transform from [xxx] to [odom]"

JaneCX, 2019-02-07:
Sorry I forgot to launch the rosbot.launch ..... soooo sorry

antopemo, 2019-02-17:
I am facing the same issue, and launching rosbot.launch didn't solved it. I am following the tutorial on the README with no luck with launching it on gazebo. Could someone help me out?

JaneCX, 2019-02-17:
> I am facing the same issue, and launching rosbot.launch didn't solved it. I am following the tutorial on the README with no luck with launching it on gazebo. Could someone help me out?

so did you notice any warning or error when you roslaunch?

lukaszmitka, 2019-02-18:
> I am facing the same issue, and launching rosbot.launch didn't solved it. I am following the tutorial on the README with no luck with launching it on gazebo. Could someone help me out?

Could you post here full output that you get from roslaunch command? It would be much easier to solve your problem.

antopemo, 2019-02-22:
Okay I'm gonna give some context over here so you can see why this may be failing.
I installed the virtual machine from Husarion Tutorials page that claims that it has everything we need to test Rosbot 2.0 on Gazebo. I installed some libraries that appeared to be missing and followed the README tutorial on this repo linked from the original tutorial page. We got to this point but on the gviz it only shows the white silhouette and gives some errors about robot publisher. 
When we launch the rosbot.launch first, it just shows the loading error that @JaneCX had at first but no publisher error this time.
Am I doing fine by following the tutorial? Cause if launching rosbot.launch was neccesary it should have been linked in the tutorial.
BTW, total noob with Gazebo or ROS here, just starting to learn so my Uni can purchase Rosbot 2.0

lukaszmitka, 2019-02-22:
Yesterday a have updated the repository, the changes include the tutorial and launch files. Could you checkout the repo and follow the updated tutorial?

closed 2019-04-08 by lukaszmitka