husarion/rosbot_ros #2: IMU joint position in rosbot

shrijan00, 2019-04-06:
Hi,

I was trying to integrate imu on ROSBOT, but the joint location to which I'm adding is incorrect due to which on doing sensor fusion using EKF, the robot starts to shake in rviz. So, could anyone help me in adding the imu at the correct location?

Hubert424, 2019-04-08:
Hi,

I've just add IMU to our model. Please check this out. If problem will be occur, I will help you to investigate it.

Best regards,
Hubert

shrijan00, 2019-04-08:
@Hubert424  Thanks for your reply. But I cannot find any imu joint available in the rosbot.xacro file neither any plugin for the imu in rosbot.gazebo.

Hubert424, 2019-04-08:
@shrijan00 IMu sensor is based inside the robot and its localization is practically the same as 'top', so basically is no need to create additional joint for it.

shrijan00, 2019-04-08:
Thanks, alot

closed 2020-06-04 by adamkrawczyk