husarion/rosbot_ros #5: Robot initial position in Rviz / gazebo mismatched

pringithub, 2019-08-05:
Hello,

Thank you for this repo! Unfortunately, every time I try to run the demo using a saved map, the rviz coordinates of the robot do not match up to the ones in gazebo (the init_pos in gazebo is always the same). Obviously, the incorrect init_pos leads to errors in localization.

[image]

lukaszmitka, 2019-08-06:
Hello,

Them map visible in Rviz screen is not exactly resembling environment from Gazebo (take a look at jersey barrier in Gazebo and three rectangles in Rviz at location of jersey barrier). The three obstacles should be continuous and form single rectangle, as long as they don't, we need to assume that map was build incorrectly. On distorted map, position will also be wrong.

Please try to create and save map again.

Regards,
Łukasz

pringithub, 2019-08-06:
Hi Lukasz,

I've tried creating and saving maps multiple times now. Every time I do this, I get either an incorrect map (a few times _very incorrect_, or an incorrect starting position. I didn't modify any of the code, and I'm running on a pretty powerful machine with an NVIDIA GPU.

lukaszmitka, 2019-08-06:
Could you then record screen showing as you are creating the map and it is getting incorrect?

adamkrawczyk, 2020-06-04:
I'm closing this issue, because of no activity.

closed 2020-06-04 by adamkrawczyk

pls30, 2020-07-02:
I have similar problems, even when driving the robot it moves in the gazebo, in rviz it remains still in its initial position. In rviz everything dances, even the wheels of the robot. I am using the lidar CPU as suggested in the tips.

Hubert424, 2020-07-13:
Please send us screenshot of Rviz window and log file from starting roslaunch.