husarion/rosbot_ros #6: Laser scan orientation

glakshay, 2019-10-24:
Hi, I want the laser to scan in vertical manner instead of horizontal so I just rotated the laser but it doesn't scan anymore, Can you tell how I can change the code so it scans in vertical direction?

Also, once i have generated a map for an environment, I am unable to scan again so for that should I delete the previous map ? If not, can you tell how to create multiple scan map files?

Lastly, while I am able to sucessfuly open the launch file in urdf and gazebo but I get thsi error in terminal:
```
`ERROR: cannot launch node of type [amcl/amcl]: amcl
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/lakshay/ros_catkin_ws/src
ROS path [2]=/home/lakshay/catkin_ws/src
ROS path [3]=/opt/ros/kinetic/share
ERROR: cannot launch node of type [move_base/move_base]: move_base
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/lakshay/ros_catkin_ws/src
ROS path [2]=/home/lakshay/catkin_ws/src
ROS path [3]=/opt/ros/kinetic/share
```
`
And this error in Rviz under maps:
```
Transform
No transform from [map] to [odom]
```

lukaszmitka, 2019-10-25:
Hello,
How did you rotate the laser? Which parameter did you change? It is not possible to stop laser from scanning by rotating it, but it could get covered by other robot elements, then all scans will be 'empty'.

What do you mean by "unable to scan again"? Do you want to add scans to existing map or create new map?

For the errors of `amcl` and `move_base`, install dependencies:
```
sudo apt install ros-kinetic-amcl ros-kinetic-move-base
```
Error `No transform from [map] to [odom]` will occur as long as `amcl` is not finding position, thus if you solve that one, this will also be solved.

glakshay, 2019-10-25:
Hi, thanks for the response. 
I just edited the urdf file- rosbot.xacro and edited rplidar Laser rpy component, but then it is unable to scan. If i want the laser to scan in vertical direction then what should I do?

What I meant by unable to scan again is that no new map is being created for the new environment.Can you tell how to add scans to existing map and also, create a new map?

Yea, I will install those dependencies.

Thanks again.

lukaszmitka, 2019-10-25:
For laser scanner in URDF model we can distinguish collision, visual, inertial and sensor plugin. Each of them is defined independently, it allows us to modularly define robot. Although, editing only one of the components will probably break model functionality.
Each component of laser scanner has the same parent, which is `rplidar_joint`. Adjust this joint to set position of laser scanner relative to ROSbot base.

New map is created each time when `gmapping` node is started, thus to create new map just start the same launch files that you used first time.
Adding scans to existing map may be more difficult, you may try to use [`multirobot_map_merge`](http://wiki.ros.org/multirobot_map_merge) for this purpose.

Regards,
Łukasz

glakshay, 2019-10-25:
From what I infer, I need to change the `rplidar_joint` in urdf file only to change the scan direction.

One more thing, If I delete all the camera parameters in urdf file so will the scan still function properly? Its cause I am trying to use this simulation for a part of my assignment and I am not using any camera so just wanted to know.

Thanks a lot Łukasz

lukaszmitka, 2019-10-25:
You can safely remove camera from the model if you do not plan on using it. Keep in mind that camera entries are both in `rosbot.gazebo` and `rosbot.xacro` files.

Regards,
Łukasz

glakshay, 2019-10-25:
Great, thank for the help.

~Lakshay

closed 2020-06-04 by adamkrawczyk