husarion/rosbot_ros #11: Rationale for Trajectory Planner max_vel?

adwiii, 2020-12-18:
In the [trajectory_planner.yaml](https://github.com/husarion/rosbot_description/blob/master/src/rosbot_navigation/config/trajectory_planner.yaml#L2) config, the maximum velocity is set as `max_vel_x: 0.1`. However it appears that the maximum speed is given [here](https://husarion.com/manuals/rosbot-manual/#rosbot-manual-overview) as 1.25 m/s or [here](https://husarion.com/manuals/rosbot/) as 1.0 m/s (separate issue, these manuals conflict). 

Is there a rationale behind using 0.1 m/s instead of 1.0 m/s?

lukaszmitka, 2020-12-18:
There is a rationale.
Value stated in manual is a maximum velocity that ROSbot can reach. The correct one is 1.0m/s
In trajectory planner set limit for velocity to 0.1 m/s as we do not want ROSbot to drive faster. But if user want to, this value may be changed.

adwiii, 2020-12-18:
Thank you. I thought 1/10th speed might be overly conservative. I will experiment with turning up the speed. Is there a reason to not want the robot to drive faster? Is it a concern with software or hardware?

lukaszmitka, 2020-12-21:
It is a safety precaution for first time users.
Both hardware and software can handle higher velocities.