husarion/rosbot_ros #12: Launching ROS2 branch in simulation

erhoye30, 2021-01-14:
I am currently having issues running the ROS2 branch of rosbot_description.  I just set up a desktop with ubuntu 20.04 and freshly installed ROS2 Foxy. I followed the instructions on your README for pulling code and was able to build both the rosbot_description and rplidar_ros.  There were some warnings regarding rplidar_ros but it was able to successfully build.  When I try to launch the demo though, the terminal seems to hang up and never successfully launches gazebo.  I have fixed some warnings regarding launch files utilizing node_executable to executable, but otherwise I have made no changes to the repository.  The output when launching the demo is below.  Please let me know if you have any advice on why this might be occurring.

$ ros2 launch rosbot_description rosbot_sim.launch.py

[INFO] [launch]: All log files can be found below /home/datavantyx/.ros/log/2021-01-14-10-34-25-943770-modern-times-74950
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [74952]
[INFO] [static_transform_publisher-2]: process started with pid [74954]
[INFO] [static_transform_publisher-3]: process started with pid [74956]
[INFO] [static_transform_publisher-4]: process started with pid [74959]
[INFO] [spawn_entity.py-5]: process started with pid [74961]
[static_transform_publisher-3] [INFO] [1610649266.249930610] [static_transform_publisher_eOqJMARNOEKdOsVO]: Spinning until killed publishing transform from 'base_link' to 'camera_link'
[static_transform_publisher-2] [INFO] [1610649266.250159112] [static_transform_publisher_SOBhqAQgzaHy6nOB]: Spinning until killed publishing transform from 'camera_link' to 'camera_depth_frame'
[static_transform_publisher-4] [INFO] [1610649266.250174536] [static_transform_publisher_6cZ7lSiFaPCHbIij]: Spinning until killed publishing transform from 'base_link' to 'laser'
[gzserver-1] Gazebo multi-robot simulator, version 11.1.0
[gzserver-1] Copyright (C) 2012 Open Source Robotics Foundation.
[gzserver-1] Released under the Apache 2 License.
[gzserver-1] http://gazebosim.org
[gzserver-1] 
[spawn_entity.py-5] [INFO] [1610649266.546508018] [spawn_entity]: Spawn Entity started
[spawn_entity.py-5] [INFO] [1610649266.546879672] [spawn_entity]: Loading entity XML from file /home/datavantyx/Documents/ros2_devel_workspace/cap/install/rosbot_description/share/rosbot_description/models/rosbot.sdf
[spawn_entity.py-5] [INFO] [1610649266.551132271] [spawn_entity]: Waiting for service /spawn_entity, timeout = 5
[spawn_entity.py-5] [INFO] [1610649266.551571638] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-5] [INFO] [1610649267.056207064] [spawn_entity]: Calling service /spawn_entity
[gzserver-1] [INFO] [1610649267.276457121] [imu_plugin]: <initial_orientation_as_reference> is unset, using default value of false to comply with REP 145 (world as orientation reference)
[gzserver-1] [INFO] [1610649267.505322972] [astra_camera]: Publishing camera info to [/camera/camera_info]
[gzserver-1] [INFO] [1610649267.506120171] [astra_camera]: Publishing depth camera info to [/camera/depth/camera_info]
[gzserver-1] [INFO] [1610649267.507336667] [astra_camera]: Publishing pointcloud to [/camera/points]
[gzserver-1] [INFO] [1610649267.529731823] [range]: missing <radiation_type>, defaulting to infrared
[gzserver-1] [WARN] [1610649267.531095107] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[gzserver-1] [INFO] [1610649267.534160022] [range]: missing <radiation_type>, defaulting to infrared
[gzserver-1] [WARN] [1610649267.535568638] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[gzserver-1] [INFO] [1610649267.539199000] [range]: missing <radiation_type>, defaulting to infrared
[gzserver-1] [WARN] [1610649267.540716006] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[gzserver-1] [Msg] Waiting for master.
[gzserver-1] [Msg] Connected to gazebo master @ http://127.0.0.1:11345
[gzserver-1] [Msg] Publicized address: 192.168.0.242
[gzserver-1] [Wrn] [msgs.cc:1841] Conversion of sensor type[depth] not supported.
[spawn_entity.py-5] [INFO] [1610649267.544146255] [spawn_entity]: Spawn status: SpawnEntity: Successfully spawned entity [rosbot]
[gzserver-1] [INFO] [1610649267.544314015] [range]: missing <radiation_type>, defaulting to infrared
[gzserver-1] [Wrn] [ColladaLoader.cc:1853] Unable to add material[None-material]
[INFO] [spawn_entity.py-5]: process has finished cleanly [pid 74961]
[gzserver-1] [INFO] [1610649267.661996308] [skid_steer_drive]: Wheel pair 1 separation set to [0.240000m]
[gzserver-1] [INFO] [1610649267.662043922] [skid_steer_drive]: Wheel pair 2 separation set to [0.240000m]
[gzserver-1] [INFO] [1610649267.662054138] [skid_steer_drive]: Wheel pair 1 diameter set to [0.080000m]
[gzserver-1] [INFO] [1610649267.662062777] [skid_steer_drive]: Wheel pair 2 diameter set to [0.080000m]
[gzserver-1] [INFO] [1610649267.665715971] [skid_steer_drive]: Subscribed to [/cmd_vel]
[gzserver-1] [INFO] [1610649267.667625861] [skid_steer_drive]: Advertise odometry on [/odom]
[gzserver-1] [INFO] [1610649267.669729468] [skid_steer_drive]: Publishing odom transforms between [odom] and [base_link]
[gzserver-1] [INFO] [1610649267.669756627] [skid_steer_drive]: Publishing wheel transforms between [base_link], [front_left_wheel_hinge] and [front_right_wheel_hinge]
[gzserver-1] [INFO] [1610649267.669764081] [skid_steer_drive]: Publishing wheel transforms between [base_link], [rear_left_wheel_hinge] and [rear_right_wheel_hinge]
[gzserver-1] [Wrn] [msgs.cc:1841] Conversion of sensor type[depth] not supported.

At this point nothing happens.

erhoye30, 2021-01-14:
I am unsure whether this was done on purpose, due to not fully understanding the rosbot_description package in ROS2, but I discovered that for running in simulation your rosbot_sim.launch.py file provided an incorrect variable.  

DeclareLaunchArgument(
            name='gui',
            default_value='false'
        ),

# Needed to be switched to 

DeclareLaunchArgument(
            name='gui',
            default_value='true'
        ),
It now launches gazebo gui as well.

delihus, 2023-05-20:
Hi @erhoye30 !
Since https://github.com/husarion/rosbot_ros/pull/43 ignition gazebo simulation is available!

Best regards,
JD

closed 2023-05-20 by delihus