husarion/rosbot_ros #47: Cannot lauch rosbot gazebo sim

LordSyd, 2023-04-10:
Hello,
I tried launching the gazebo sim from the humble branch and always get the following error:

`INFO] [launch]: All log files can be found below /home/fhcampus01/.ros/log/2023-04-10-15-01-24-096082-fh-campus-wien-01-9467
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: "package 'husarion_office_gz' not found, searching: ['/opt/ros/humble', '/home/fhcampus01/Documents/repos/rosbot/rosbot_ros/install/rosbot_gazebo', '/home/fhcampus01/Documents/repos/rosbot/rosbot_ros/install/rosbot', '/home/fhcampus01/Documents/repos/rosbot/rosbot_ros/install/rosbot_bringup', '/home/fhcampus01/Documents/repos/rosbot/rosbot_ros/install/rosbot_controller', '/home/fhcampus01/Documents/repos/rosbot/rosbot_ros/install/rosbot_description']"`

Am I doing something wrong here is this a bug in the package?

delihus, 2023-05-04:
Hi @LordSyd.
I will update README.md soon.
Try this:
Install vcs like a python package:
`apt install  python3-vcstool'
and then download the repositories from . repos files e.g.:
```
vcs import src < src/rosbot/rosbot_hardware.repos
vcs import src < src/rosbot/rosbot_simulation.repos
```

Then build the project.

Best regards,
JD

delihus, 2023-05-10:
Instruction added here  https://github.com/husarion/rosbot_ros/pull/49

closed 2023-05-10 by delihus