husarion/rosbot_ros #48: Running Ros2 Humble simulation for the  Rosbot with ignition-gazebo

dantepiotto, 2023-04-27:
I have been unsuccessfully trying to run an ignition-gazebo simulation of the rosbot and would like to ask for assistance. In my workspace I have succesfully built the following packages:
-ros_components_description
-rosbot_hardware_interfaces
-ign_ros2_control
-husarion_office_gz
-ign_ros2_control_demos
-rosbot_description
-rosbot_controller
-rosbot_bringup
-rosbot
-rosbot-gazebo
I have tried launching I have changed a couple of the default values of launch arguments in the launch files bringup.launch.py and controller.launch.py as such:
-use_sim: True
-simulation_engine: ignition-gazebo

I have tried this both with the main branch and the add-mecanum-wheels branch but got the same result with both: once I ros2 launch rosbot_bringup bringup.launch.py, I get the following message:
[spawner-2] [INFO] [1682588256.303546245] [spawner_joint_state_broadcaster]: Waiting for '/simulation_controller_manager' node to exist

and nothing else happens

I would appreciate any input or suggestions on what I might be getting wrong

delihus, 2023-05-03:
Hi @dantepiotto!
Bringing up simulation is not described yet!
I will add tutorial in README. To bringup the simulation use:
```bash
ros2 launch rosbot_gazebo simulation.launch.py
```

If you have still problems please reopen the issue.

JD

ma-shangao, 2023-07-14:
Hi @delihus ,
I am following the Build and run Gazebo simulation in README and I tried the
`ros2 launch rosbot_gazebo simulation.launch.py`,
but still get similar problem:
`[spawner-5] [INFO] [1689360357.792700055] [spawner_joint_state_broadcaster]: Waiting for '/simulation_controller_manager' node to exist`.

I am currently using ROS 2 humble and GazeboSim Garden. Any suggestions? Thank you very much.

delihus, 2023-07-16:
@ma-shangao Please send me all logs.
JD

ma-shangao, 2023-07-17:
> @ma-shangao Please send me all logs. JD

Hi @delihus, 
Sure, here are the logs:

## create_***.log
```
[INFO] [1689552553.812915972] [ros_gz_sim]: Requesting list of world names.
[INFO] [1689552554.260030625] [ros_gz_sim]: Waiting messages on topic [robot_description].
[INFO] [1689552554.359101431] [ros_gz_sim]: Requested creation of entity.
[INFO] [1689552554.359194189] [ros_gz_sim]: OK creation of entity.
```

## ekf_node_***.log
```[INFO] [1689552565.715116483] [rclcpp]: signal_handler(signum=2)```

## parameter_bridge_***.log
```
[INFO] [1689552553.832276946] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/scan (ignition.msgs.LaserScan) -> /scan (sensor_msgs/msg/LaserScan)] (Lazy 0)
[INFO] [1689552553.835468153] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/camera/camera_info (ignition.msgs.CameraInfo) -> /camera/camera_info (sensor_msgs/msg/CameraInfo)] (Lazy 0)
[INFO] [1689552553.837061691] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/camera/depth_image (ignition.msgs.Image) -> /camera/depth_image (sensor_msgs/msg/Image)] (Lazy 0)
[INFO] [1689552553.837952780] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/camera/image (ignition.msgs.Image) -> /camera/image (sensor_msgs/msg/Image)] (Lazy 0)
[INFO] [1689552553.838485953] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/camera/points (ignition.msgs.PointCloudPacked) -> /camera/points (sensor_msgs/msg/PointCloud2)] (Lazy 0)
[INFO] [1689552553.839391526] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[INFO] [1689552565.715121211] [rclcpp]: signal_handler(signum=2)
```

## python3.10_***.log
```
[INFO] [1689552556.034787962] [spawner_joint_state_broadcaster]: Waiting for '/simulation_controller_manager' node to exist
[INFO] [1689552558.042677897] [spawner_joint_state_broadcaster]: Waiting for '/simulation_controller_manager' node to exist
[INFO] [1689552560.051540036] [spawner_joint_state_broadcaster]: Waiting for '/simulation_controller_manager' node to exist
[INFO] [1689552562.059081923] [spawner_joint_state_broadcaster]: Waiting for '/simulation_controller_manager' node to exist
[INFO] [1689552564.066695516] [spawner_joint_state_broadcaster]: Waiting for '/simulation_controller_manager' node to exist
```

## robot_state_publisher_***.log
```
[INFO] [1689552553.843996557] [robot_state_publisher]: got segment base_link
[INFO] [1689552553.844083282] [robot_state_publisher]: got segment body_link
[INFO] [1689552553.844095840] [robot_state_publisher]: got segment camera_link
[INFO] [1689552553.844099653] [robot_state_publisher]: got segment cover_link
[INFO] [1689552553.844102941] [robot_state_publisher]: got segment depth
[INFO] [1689552553.844106324] [robot_state_publisher]: got segment fl_range
[INFO] [1689552553.844109672] [robot_state_publisher]: got segment fl_wheel_link
[INFO] [1689552553.844113034] [robot_state_publisher]: got segment fr_range
[INFO] [1689552553.844116091] [robot_state_publisher]: got segment fr_wheel_link
[INFO] [1689552553.844119051] [robot_state_publisher]: got segment imu_link
[INFO] [1689552553.844122219] [robot_state_publisher]: got segment laser
[INFO] [1689552553.844125236] [robot_state_publisher]: got segment orbbec_astra_link
[INFO] [1689552553.844128557] [robot_state_publisher]: got segment rl_range
[INFO] [1689552553.844131546] [robot_state_publisher]: got segment rl_wheel_link
[INFO] [1689552553.844134540] [robot_state_publisher]: got segment rr_range
[INFO] [1689552553.844137473] [robot_state_publisher]: got segment rr_wheel_link
[INFO] [1689552553.844140523] [robot_state_publisher]: got segment slamtec_rplidar_a2_link
[INFO] [1689552565.715118986] [rclcpp]: signal_handler(signum=2)
```
Thank you for your help.

ma-shangao, 2023-07-24:
Hi @delihus,
I think I have solved this problem. It is probably due to my Gazebo version. I tried to remove Gazebo garden and reinstall Gazebo fortress. Now everything works fine. Thank you for your time.

closed 2023-07-31 by delihus