husarion/rosbot_ros #51: rosbot_bringup failure

jnaorgm, 2023-06-19:
Hi,

After building the repo and trying to run rosbot_bringup.launch.py i get the following error:

package not found: "package 'ros_components_description' not found

What am I doing wrong?

Thanks

jnaorgm, 2023-06-19:
Thanks!

Runs now, but teleop doesn't do anything on the robot. In the logs there are messages like these:

[ros2_control_node-1] [WARN] [1687185819.543073517] [RosbotImuSensor]: Feedback message from imu wasn't received yet
[ros2_control_node-1] [WARN] [1687185820.043386661] [RosbotImuSensor]: Feedback message from imu wasn't received yet
[ros2_control_node-1] [WARN] [1687185820.543622713] [RosbotImuSensor]: Feedback message from imu wasn't received yet
[ros2_control_node-1] [WARN] [1687185821.043846154] [RosbotImuSensor]: Feedback message from imu wasn't received yet
[ros2_control_node-1] [WARN] [1687185821.544182705] [RosbotImuSensor]: Feedback message from imu wasn't received yet
[ros2_control_node-1] [WARN] [1687185822.044495293] [RosbotImuSensor]: Feedback message from imu wasn't received yet
[ros2_control_node-1] [WARN] [1687185822.544829085] [RosbotImuSensor]: Feedback message from imu wasn't received yet
[spawner-3] [INFO] [1687185822.829340139] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available

delihus, 2023-06-19:
Did you update the firmware? 
Follow this page: https://husarion.com/tutorials/howtostart/rosbot2r-quick-start/

This command upgrades the firmware to the latest version:
```
docker run \
--rm -it --privileged \
husarion/rosbot:humble \
/flash-firmware.py /root/firmware.bin
```

delihus, 2023-07-31:
I'm closing due to inactivity. If the issue is still valid, please reopen it.

closed 2023-07-31 by delihus