husarion/rosbot_ros #53: Problem launching Gazebo from docker

julietcookies, 2023-06-27:
We followed the guidelines specified on your website (i.e., running the Gazebo docker and then running the ros2 launch command on the simulation.launch.py file). Unfortunately, Gazebo doesn’t fully launch since we receive an error specifying that the simulation_launch_node doesn’t exist. Our current solution to this is to run a compose script, running multiple images (including a bridge ros1 and ros2 image) and run with ros1. This makeshift solution seems like an overkill. Has anyone run into this issue? If so, how do we correct this?
Some more important notes:
** We are working with Husarion’s Rosbot 2R. 
** We are working with ros2.
** We are running this on a docker with a Gazebo image imported from the Husarion website

delihus, 2023-06-27:
Hi @julietcookies!
If you have ROS 2 Humble installed natively you can follow this instruction https://github.com/husarion/rosbot_ros#build-and-run-gazebo-simulation.

If you want to use docker look at this demo: https://github.com/husarion/rosbot-docker/blob/ros2/demo/compose.simulation.yaml

Please tell me if this solved your issues!

Best regards,
JD

julietcookies, 2023-06-28:
Thank you for the quick response. Unfortunately when following the instructions given in the GitHub, we encounter the same issue - "Waiting for '/simulation_controller_manager' node to exist". The same issue happens when running the docker demo (i.e., running the compose file given and launching gazebo through it).

Is there another solution to combat this (specifically the simulation_controller_manager node missing)?

delihus, 2023-06-28:
Please send me whole logs.

Heenne, 2023-08-16:
I would like to reopen this issue as I have the same problem.
Currently I am trying to get the ROSbot 2 in the simulation to work.

I get the following log in the console:
[2023_08_16_simulation_control_manager_node_not_existing.txt](https://github.com/husarion/rosbot_ros/files/12362171/2023_08_16_simulation_control_manager_node_not_existing.txt)

Maybe this is a little bit off-topic, but would you recommend Gazebo-Classic or Gazebo-Sim (Fortress or Garden) to use with the ROSbot? I am currently setting things up for a lab at out university.

delihus, 2023-08-17:
Hi @Heenne,
we are using the Fortress version of Gazebo-Sim. 
Please send the output of:
```
ign gazebo --versions
```

Best regards,
JD

Heenne, 2023-08-17:
Hi @delihus 
thanks for the quick reply. This is the output:
`
Singularity> ign gazebo --version
Ignition Gazebo, version 6.11.0
Copyright (C) 2018 Open Source Robotics Foundation.
Released under the Apache 2.0 License.
`

Best regards
Heenne

delihus, 2023-08-17:
I have just added the requirements before running the simulation. Please see https://github.com/husarion/rosbot_ros#build-and-run-gazebo-simulation.
Try export this global variable:
 ```bash
 export GZ_VERSION=fortress
 ```

Heenne, 2023-08-17:
I executed the command and launched the simulation.launch.py again, but nothing changed.
In the log there is shown that the GZ_VERSION contains "fortress" and the error remains.

[2023_08_17_simulation_control_manager_node_not_existing.txt](https://github.com/husarion/rosbot_ros/files/12367116/2023_08_17_simulation_control_manager_node_not_existing.txt)

Heenne, 2023-08-17:
The path to ign_ros2_control is `/opt/ros/humble/ `
The version of the ign_ros2_control is 0.7.0. 
First I used the "vcs"-command in the container itself. Now I inserted it into the definition-file of the container that is used for the building process. That worked.
So now the simulation works in Gazebo Fortress. 

Just a heads up: The path to the .repos-file is incorrect. It should be `src/rosbot_ros/rosbot/rosbot_simulation.repos`
Also it might be a good idea to include an information that one must cd into the workspace folder.

Thank you very much for the help and the well documented repository. Very looking forward to the real robot (will be ordered soon).

delihus, 2023-08-18:
It's great that you solved the issue.

> First I used the "vcs"-command in the container itself. Now I inserted it into the definition-file of the container that is used for the building process. That worked.
You can check our docker images with the simulation in the [rosbot-docker](https://github.com/husarion/rosbot-docker) repository.

> Just a heads up: The path to the .repos-file is incorrect. It should be src/rosbot_ros/rosbot/rosbot_simulation.repos
Also it might be a good idea to include an information that one must cd into the workspace folder.

In the instruction you have to clone repository into the `src` folder.
```
git clone https://github.com/husarion/rosbot_ros src/
```

Best regards,
JD

closed 2023-08-18 by delihus