husarion/rosbot_ros #54: Issue seeing topics in topic list ROS1 vs ROS2

julietcookies, 2023-06-28:
We are currently running Gazebo through running the following .yaml file in a composed (i.e., running multiple images simultaneously) fashion: 

services:
  ros-master:
      image: ros:noetic-ros-core
  #    network_mode: host
      command: stdbuf -o L roscore
      
    bridge:
      image: husarion/ros:galactic-ros1-bridge
  #    network_mode: host
      environment:
        - ROS_MASTER_URI=http://ros-master:11311
      command: ros2 run ros1_bridge dynamic_bridge
  
    rosbot_simulation:
      image: husarion/rosbot:noetic-simulation
  #    network_mode: host
      volumes:
        - /tmp/.X11-unix:/tmp/.X11-unix:rw
        - $HOME:$HOME
      environment:
        - DISPLAY=${DISPLAY:?err}
        - LIBGL_ALWAYS_SOFTWARE=1
        - ROS_MASTER_URI=http://ros-master:11311
      command: >
        roslaunch --wait rosbot_bringup rosbot.launch 
          world:=/ros_ws/src/rosbot_gazebo/worlds/delete.world    

When attaching to the different dockers running, we can only see the topics published by the simulated robot in the ROS1 images. Meaning, we are unable to see the topics published in any of the ROS2 oriented images. This is a problem given that we would like to run ROS2 toolboxes, but given that the topics published by the simulated robot do not appear in ROS2 images, when attaching to these dockers, we are unable to run the toolboxes on the relevant information.

delihus, 2023-06-28:
Please try and run the simulation as I described here https://github.com/husarion/rosbot_ros/issues/53#issuecomment-1609023247.

Best regards,
JD

delihus, 2023-07-31:
I'm closing due to inactivity. If the issue is still valid, please reopen it.

closed 2023-07-31 by delihus