husarion/rosbot_ros #55: missing link for depth camera in simulation, broken robot_description

Neathle, 2023-06-29:
I am simulating the ROSBOT2 in humble, using the docker image provided, and using the simulation.launch.py provided as well in the repo. When I launch the simulation, everything works correctly, except the depth camera, which appears to be missing a link to the rest of the robot. 
The depth camera has created its own tree: /rosbot/base_link/orbecc_astra_camera. Instead of using the depth link to publish. 
[image]
[image]

Robot_description is also broken, as it can be seen in the image. I tried with rviz2 as well and it simply doesn't display the robot.
I solved the issue by manually linking both frames in the /tf_static topic, but this should be corrected.
I include the generated urdf from doing ```xacro /ros2_ws/src/rosbot_description/urdf/rosbot.urdf.xacro```:
```urdf
<?xml version="1.0" ?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from /ros2_ws/src/rosbot_description/urdf/rosbot.urdf.xacro | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<robot name="rosbot">
  <link name="base_link"/>
  <joint name="base_to_body_joint" type="fixed">
    <origin rpy="0.0 0.0 0.0" xyz="0.0 0.0 0.0425"/>
    <parent link="base_link"/>
    <child link="body_link"/>
  </joint>
  <link name="body_link">
    <visual>
      <geometry>
        <mesh filename="file:///ros2_ws/install/rosbot_description/share/rosbot_description/meshes/body.stl" scale="0.001 0.001 0.001"/>
      </geometry>
      <origin rpy="1.5707963267948966 0.0 1.5707963267948966" xyz="0.0 0.0 -0.0173"/>
      <material name="White">
        <color rgba="1.0 1.0 1.0 1.0"/>
      </material>
    </visual>
    <collision>
      <geometry>
        <box size="0.197 0.150 0.080"/>
      </geometry>
      <origin rpy="0.0 0.0 0.0" xyz="0.0 0.0 0.02"/>
    </collision>
    <inertial>
      <mass value="1.728"/>
      <inertia ixx="0.004162" ixy="0.0" ixz="0.0" iyy="0.00651" iyz="0.0" izz="0.008828"/>
      <origin rpy="0.0 0.0 0.0" xyz="0.0 0.0 0.02"/>
    </inertial>
  </link>
  <gazebo reference="body_link">
    <material>Gazebo/White</material>
  </gazebo>
  <joint name="body_to_cover_joint" type="fixed">
    <origin rpy="0.0 0.0 0.0" xyz="0.0 0.0 0.0603"/>
    <parent link="body_link"/>
    <child link="cover_link"/>
  </joint>
  <link name="cover_link">
    <visual>
      <geometry>
        <mesh filename="file:///ros2_ws/install/rosbot_description/share/rosbot_description/meshes/cover.stl" scale="0.001 0.001 0.001"/>
      </geometry>
      <origin rpy="1.5707963267948966 0.0 1.5707963267948966" xyz="0.0 0.0 -0.002"/>
      <material name="Red">
        <color rgba="0.8 0.0 0.0 1.0"/>
      </material>
    </visual>
    <gazebo reference="cover_link">
      <material>Gazebo/Red</material>
    </gazebo>
  </link>
  <joint name="body_to_imu_joint" type="fixed">
    <origin rpy="0.0 0.0 0.0" xyz="0.003 -0.0495 0.04332"/>
    <parent link="body_link"/>
    <child link="imu_link"/>
  </joint>
[... 285 lines cut ...]
      <geometry>
        <box size="0.04 0.165 0.03"/>
      </geometry>
    </collision>
    <!-- mounting base model collision -->
    <collision>
      <origin rpy="0.0 0.0 0.0" xyz="-0.006499999999999999 0.0 0.0095"/>
      <geometry>
        <box size="0.06 0.06 0.019"/>
      </geometry>
    </collision>
    <inertial>
      <origin rpy="0.0 0.0 0.0" xyz="0.0 0.0 0.028999999999999998"/>
      <mass value="0.300"/>
      <inertia ixx="0.00073916" ixy="0.0" ixz="0.0" iyy="0.00015435" iyz="0.0" izz="0.00077395"/>
    </inertial>
  </link>
  <joint name="orbbec_astra_to_depth_joint" type="fixed">
    <origin rpy="-1.5707963267948966 0.0 -1.5707963267948966" xyz="0.01 0.0 0.035"/>
    <parent link="orbbec_astra_link"/>
    <child link="depth"/>
  </joint>
  <link name="depth"/>
</robot>
```

Neathle, 2023-07-05:
Thank you very much! I have switched back to rviz for now, but I will try that fix to see if foxglove is viable. The problem with the depth camera is still there even in rviz.

delihus, 2023-07-11:
Hi @Neathle,
I looked at the issue and it seems like this problem https://github.com/gazebosim/gz-sensors/issues/239. It is on gazebo side and as I see they fixed this here https://github.com/gazebosim/gz-sensors/pull/350 but only for `gz-sensors7 ` but we are using the `ign-sensors6`.

I made propose workaround for now here https://github.com/husarion/rosbot_ros/tree/ros2-ign-camera-fix. 
[image]
As I see the rviz2 has issues with display the depth image in this format...
[image]
But `rqt_image_viewer` works perfect...
[image]

Please inform me if the solution works for you and feel free to reach out if you need any further assistance.
Best regards,
JD

delihus, 2023-07-31:
I'm closing due to inactivity. If the issue is still valid, please reopen it.

closed 2023-07-31 by delihus