husarion/rosbot_ros #65: Noetic Simulation

CKarlGrech, 2023-08-25:
I followed the installation process for the simulations in Noetic, however when I run the final "catkin cmake" command, the workspace is not built as it says that the folders: "rosbot_bringup", "rosbot_controller", "rosbot description", "rosbot" and "rosbot_gazebo" are homegenous folders.

How can I solve this issue please?

Thank you

rafal-gorecki, 2023-08-28:
Hello @CKarlGrech,
From the information I can see, it's hard for me to pinpoint the cause of the problem. Below are some suggestions that may help you solve the problem.
1. I would ask you to check if the branch version is compatible with your ROS version.
2. The correct build command is `catkin_make` (not `catkin cmake`.
3. Try removing the content of the workspace, add the src file and build an empty workspace.

_Best regards,
RG_

CBeaune, 2023-09-20:
I had the same issue as @CKarlGrech 
My problem was that, when cloning the repo on local, the branch is still on the default one (humble) and thus cannot build the package as it does not correspond to a ROS1 one. 
Try:
 `cd ~/ros_workspace/src/rosbot_ros
git checkout noetic
`
Then you can build the noetic branch of the repo. 
Best,

delihus, 2023-10-04:
Thank you @CBeaune .

We have updated the readme!
https://github.com/husarion/rosbot_ros/pull/72
Best,
JD

closed 2023-10-04 by delihus

CKarlGrech, 2023-10-30:
Thank you both for the help