husarion/rosbot_ros #77: Depth Camera Limit

farishajdarpasic, 2023-10-27:
Hello. While I was working with depth images that I got from this robot in ROS, I realized, depth is limited to about 5m. Is there any way to make this limit larger (or make it limitless)? Orbec Astra camera is used (the one from the URDF file in repo, I did not change anything).

rafal-gorecki, 2023-11-08:
Hello and sorry you had to wait so long for a reply.

If you are working with a physical camera, it is not possible to increase this range as much as you expect. The Astra camera is a fairly cheap depth camera, but its capabilities are limited.

However, if you are working in a simulation, you can freely set the camera parameters. To do this, you should go to the ros_component_descriptions package and replace the following values ​​in the [astra](https://github.com/husarion/ros_components_description/blob/ros2/urdf/orbbec_astra.urdf.xacro) file:

```xml
    <clip>
      <near>0.6</near>
      <far>8.0</far>
    </clip>
```

Regards

farishajdarpasic, 2023-11-11:
What does <clip>, <near> and <far> mean? Does <far> means depth? I am working only in simulation, I am not using physical camera. Thanks

rafal-gorecki, 2023-11-11:
Gazebo use parameters in this file to create depth camera. `near` and `far` describe min and max distance returned by depth camera. After change remember to rebuild code. I also remember that rviz have trouble to decode and display this depth image, but it works so dont you worry about that. You can also have trouble to see the most distance obj on visualization, because is so dark.

farishajdarpasic, 2023-11-11:
Thank you very much!

closed 2023-11-11 by farishajdarpasic