husarion/rosbot_ros #102: Cannot flash new firmware

PhilShaw852, 2024-02-15:
With the command as detailed in the procedure:

ros2 launch rosbot_utils flash_firmware

The response below is seen:

file 'flash_firmware' was not found in the share directory of package 'rosbot_utils' which is at '/home/husarion/ros2_ws/install/rosbot_utils/share/rosbot_utils'

Any suggestions?

rafal-gorecki, 2024-02-15:
Hi @PhilShaw852, 
Thank you for the information. We made a typo, because we run this command via `ros2 run`.
Let me know if it helped.

PhilShaw852, 2024-02-15:
Thanks Rafal, I now get:
ros2 run rosbot_utils flash_firmware

Flashing /home/husarion/ros2_ws/install/rosbot_utils/share/rosbot_utils/firmware/firmware-0.11.0.bin over /dev/ttyUSB0
System architecture: aarch64
Device: RPi

Traceback (most recent call last):
  File "/home/husarion/ros2_ws/install/rosbot_utils/lib/python3.10/site-packages/rosbot_utils/flash-firmware.py", line 136, in <module>
    main()
  File "/home/husarion/ros2_ws/install/rosbot_utils/lib/python3.10/site-packages/rosbot_utils/flash-firmware.py", line 131, in main
    flasher.flash_firmware()
  File "/home/husarion/ros2_ws/install/rosbot_utils/lib/python3.10/site-packages/rosbot_utils/flash-firmware.py", line 102, in flash_firmware
    self.try_flash_operation("Write-UnProtection", sh.stm32flash, ["-u"])
  File "/usr/lib/python3/dist-packages/sh.py", line 3518, in __getattr__
    return self.__env[name]
  File "/usr/lib/python3/dist-packages/sh.py", line 3316, in __getitem__
    raise CommandNotFound(k)
sh.CommandNotFound: stm32flash
Firmware flashing failed with return code 1

rafal-gorecki, 2024-02-15:
You need also install `stm32flash`, use `sudo apt install stm32flash`.

PhilShaw852, 2024-02-16:
Unfortunately I think there is still an issue. I have also now lost the ability to teleop the robot.

husarion@my-robot:~/ros2_ws$ ros2 run rosbot_utils flash_firmware
Flashing /home/husarion/ros2_ws/install/rosbot_utils/share/rosbot_utils/firmware/firmware-0.11.0.bin over /dev/ttyUSB0
System architecture: aarch64
Device: RPi

stm32flash 0.5

http://stm32flash.sourceforge.net/

Interface serial_posix: 57600 8E1
Version      : 0x31
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0413 (STM32F40xxx/41xxx)
- RAM        : Up to 128KiB  (12288b reserved by bootloader)
- Flash      : Up to 1024KiB (size first sector: 1x16384)
- Option RAM : 16b
- System RAM : 30KiB
Write-unprotecting flash

Write-UnProtection error! Trying again.
Error: 

  RAN: /usr/bin/stm32flash /dev/ttyAMA0 -u

  STDOUT:

  STDERR:
Failed to read ACK byte
Failed to write-unprotect flash

---------------------------------------
stm32flash 0.5

http://stm32flash.sourceforge.net/

Interface serial_posix: 57600 8E1

Write-UnProtection error! Trying again.
Error: 

  RAN: /usr/bin/stm32flash /dev/ttyAMA0 -u

  STDOUT:

  STDERR:
Failed to read ACK byte

---------------------------------------
stm32flash 0.5

http://stm32flash.sourceforge.net/

Interface serial_posix: 57600 8E1

Write-UnProtection error! Trying again.
Error: 

  RAN: /usr/bin/stm32flash /dev/ttyAMA0 -u

  STDOUT:

  STDERR:

---------------------------------------
WARNING! Write-UnProtection went wrong.
stm32flash 0.5

http://stm32flash.sourceforge.net/

Interface serial_posix: 57600 8E1

Read-UnProtection error! Trying again.
Error: 

  RAN: /usr/bin/stm32flash /dev/ttyAMA0 -k

  STDOUT:

  STDERR:
Failed to read ACK byte
Unexpected reply from device on command 0x01

---------------------------------------
stm32flash 0.5

http://stm32flash.sourceforge.net/

Interface serial_posix: 57600 8E1
Version      : 0x31
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0413 (STM32F40xxx/41xxx)
- RAM        : Up to 128KiB  (12288b reserved by bootloader)
- Flash      : Up to 1024KiB (size first sector: 1x16384)
- Option RAM : 16b
- System RAM : 30KiB
Read-UnProtecting flash
Done.

stm32flash 0.5

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Interface serial_posix: 115200 8E1

Flashing error! Trying again.
Error: 

  RAN: /usr/bin/stm32flash /dev/ttyAMA0 -v -w /home/husarion/ros2_ws/install/rosbot_utils/share/rosbot_utils/firmware/firmware-0.11.0.bin -b 115200

  STDOUT:

  STDERR:
Failed to init device.

---------------------------------------
stm32flash 0.5

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Interface serial_posix: 115200 8E1
Version      : 0x31
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0413 (STM32F40xxx/41xxx)
- RAM        : Up to 128KiB  (12288b reserved by bootloader)
- Flash      : Up to 1024KiB (size first sector: 1x16384)
- Option RAM : 16b
- System RAM : 30KiB
Write to memory
Erasing memory

Flashing error! Trying again.
Error: 

  RAN: /usr/bin/stm32flash /dev/ttyAMA0 -v -w /home/husarion/ros2_ws/install/rosbot_utils/share/rosbot_utils/firmware/firmware-0.11.0.bin -b 115200

  STDOUT:

  STDERR:
Failed to read memory at address 0x08000000

---------------------------------------
stm32flash 0.5

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Interface serial_posix: 115200 8E1

Flashing error! Trying again.
Error: 

  RAN: /usr/bin/stm32flash /dev/ttyAMA0 -v -w /home/husarion/ros2_ws/install/rosbot_utils/share/rosbot_utils/firmware/firmware-0.11.0.bin -b 115200

  STDOUT:

  STDERR:
Failed to read ACK byte

---------------------------------------
WARNING! Flashing went wrong.
Done!
Firmware flashing completed successfully.

rafal-gorecki, 2024-02-16:
Please make sure that no programs using the firmware are running in the background, as it is not possible to use the port when it is already occupied.
1. Check with the `docker ps -a` command whether there are any running processes. If there are any, please remove them.
2. Disable all ros2 processes. You can check them via `ros2 node list`

If this doesn't work, try using the `~/flash_firmware.sh` script. If this script does not exist, I recommend updating the system to the latest Husarion OS
https://husarion.com/software/os/installation/.

closed 2024-05-08 by rafal-gorecki