husarion/rosbot_ros #104: migration from ignition

NarimanNiknejad, 2024-02-23:
Hi,

I was trying to use the available code you have with some plugins that we wrote and realized that the supported controller is written for ignition and not Gazebo sim. I was wondering if you guys have any plans in the near future to port the code for Gazebo sim. The following link might be useful.

https://gazebosim.org/docs/garden/migration_from_ignition

Thank you so much for your time in advance. If this happens, it is very helpful for our research.

rafal-gorecki, 2024-02-26:
Hello @NarimanNiknejad,

The article you sent really only concerns the issue of naming, where the creators, after several years of work with Ignition, return to the original name Gazebo. On this version of Gazebo (the recent Igntion) our simulation is set.

So I guess you didn't mean what the article indicates, you're just writing about Gazebo Classic. If I'm wrong, please explain what you meant when you wrote _"Gazebo sim"_.

**ROSbot + Gazebo Classic**
If you mean Gazebo Classic, known from ROS1, we do not plan to make any major changes, because Gazebo Classic ends its life in 2025. Moreover, for ROS2 it is recommended to use the newer version of Gazebo Fortress - the old Ignition Fortress.

If you want to use an older version, you can try changing the value of `simulation_engine` in `simulation.launch.py` to `gazebo-classic'`, but this is a historical thing from when we moved from ROS1 to ROS2, so it may turn out that this parameter does not fully work.

NarimanNiknejad, 2024-02-26:
Hi @rafal-gorecki,

Thank you for your time and response. I wanted to use the simulation in Gazebo Garden, but I could not. My guess was that it is because we need to set the global parameter for the simulation to Fortress. I was wondering if there are any plans to make the simulation available for Gazebo Garden.

Thank you!

rafal-gorecki, 2024-02-26:
Hi @NarimanNiknejad,

We are rather focusing on LTS versions and most likely with the release of ROS Jazzy (LTS), we will upgrade our code to be compatible with GZ Harmonic (LTS). So there is probably no direct support for GZ Garden.

NarimanNiknejad, 2024-03-17:
Hi @rafal-gorecki,

thank you so much for your kind response! I look forward ti it!

closed 2024-03-18 by rafal-gorecki