husarion/rosbot_ros #110: Invalid parameter "use_gpu"

ambitious-octopus, 2024-06-12:
Keep getting this error when running:
`ros2 launch rosbot_gazebo simulation.launch.py`

```bash
[INFO] [launch.user]: Launching namespace= init_pose={'x': '0.0', 'y': '2.0', 'z': '0.0', 'roll': '0.0', 'pitch': '0.0', 'yaw': '0.0'}
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executed command failed. Command: /opt/ros/humble/bin/xacro /root/ros_workspace/install/rosbot_description/share/rosbot_description/urdf/rosbot.urdf.xacro mecanum:=False use_sim:=True use_gpu:=True simulation_engine:=ignition-gazebo namespace:=
Captured stderr output: error: Invalid parameter "use_gpu"
when instantiating macro: slamtec_rplidar_a2 (/root/ros_workspace/install/ros_components_description/share/ros_components_description/urdf/slamtec_rplidar_a2.urdf.xacro)
in file: /root/ros_workspace/install/rosbot_description/share/rosbot_description/urdf/rosbot.urdf.xacro
```

rafal-gorecki, 2024-06-12:
Hello @ambitious-octopus,

Can you remove your workspace and rebuild rosbot_ros repository according to `README.md`. 
Let me know if anything has changed.

ambitious-octopus, 2024-06-13:
Solved by removing args in colcon build:
https://github.com/husarion/rosbot_ros/blob/9eed2811a3652e562afa9180a2db79b209d52b95/tools/Dockerfile.dev#L34-L36

```
 RUN MYDISTRO=${PREFIX:-ros}; MYDISTRO=${MYDISTRO//-/} && \ 
     source /opt/$MYDISTRO/$ROS_DISTRO/setup.bash && \ 
     colcon build
```

rafal-gorecki, 2024-06-13:
I'm happy it works, closing. ;)