husarion/rosbot_ros #130: Installation problem ROS2 Jazzy Ubunto 24.04

MF-Ahmed, 2025-03-21:
Hello, 

When configuring for simulation after 
`rosdep` install -i --from-path src --rosdistro jazzy -y `` 

I get the following dependency issues:

``WARNING: Skipping /usr/lib/python3.12/dist-packages/pybind11-2.11.1.dist-info due to invalid metadata entry 'name'
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
rosbot_controller: Cannot locate rosdep definition for [rosbot_hardware_interfaces]
rosbot_bringup: Cannot locate rosdep definition for [micro_ros_agent] ``

Thank you

delihus, 2025-03-24:
Thank you for requesting this problem.

The problem is that there is no `condition="$HUSARION_ROS_BUILD_TYPE == hardware"` in for the depended packages in  the file`rosbot_bringup/package.xml`.
We are already working on it. The bug will be resolved in the next release https://github.com/husarion/rosbot_ros/pull/127

For now you can apply this changes https://github.com/husarion/rosbot_ros/blob/0c46e855dc0e01c5804dcdd0f44307c9552031a0/rosbot_bringup/package.xml to your `rosbot_bringup/package.xml` and then run `rosdep` once again.

rafal-gorecki, 2025-06-06:
Should be resolved

closed 2025-06-06 by rafal-gorecki