husarion/rosbot_ros #131: Issue with Camera Plugin in Gazebo Simulation for ROSbot on ROS Jazzy

Tumucin, 2025-04-01:
First of all, thank you for your excellent work on this project.

I have successfully installed and run the ROSbot in a Gazebo simulation. I am able to receive IMU messages and control the ROSbot without any issues. However, I am not receiving any data on the camera topic.

I came across the following discussion:
(https://community.husarion.com/t/rosbot-2r-enabling-camera-in-gazebo-simulation/1995/5)

It mentions that the camera plugin is not compatible with ROS Jazzy and suggests a different ROS distribution. However, I would prefer to continue using the officially recommended versions of ROS, Gazebo, and Ubuntu.

Could you please provide a formal solution or guidance on how to enable the camera with Jazzy?

rafal-gorecki, 2025-04-07:
On the jazzy branch we haven't supported camera data transfer yet. However, I've just merged a bigger task with the merge of ROSbot XL with ROSbot 3, which should also add the camera data topic. 

Since there have been a lot of changes in the rep, I recommend creating a new wokspace and reading the README. 

Let me know if there are any problems. If it's OK, please close the issue.

Tumucin, 2025-04-08:
Now it works.
Thank you.

closed 2025-04-08 by Tumucin