husarion/rosbot_ros #143: TypeError after performing apt upgrade: Failed to normalize given item of type '<class 'tuple'>', when only 'str' or 'launch.Substitution' were expected.

dfavr24, 2025-07-04:
```
husarion@my-ROSbotXL:~/rosbot_ws_jazzy$ ros2 launch rosbot_bringup bringup.launch.py robot_model:=rosbot_xl -d
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[INFO] [launch]: All log files can be found below /home/husarion/.ros/log/2025-07-04-16-35-15-657721-my-ROSbotXL-280903
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7e8c34c1ce90>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7e8c34c1ce90>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7e8c37fb8fe0>'
Executing <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/jazzy/lib/python3.12/site-packages/launch/launch_service.py:228> result=None created at /opt/ros/jazzy/lib/python3.12/site-packages/launch/launch_service.py:319> took 0.463 seconds
[INFO] [robot_state_publisher-1]: process started with pid [280907]
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessStarted'
[INFO] [ros2_control_node-2]: process started with pid [280908]
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessStarted'
[INFO] [micro_ros_agent-3]: process started with pid [280909]
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStdout'
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessStarted'
[INFO] [ekf_node-4]: process started with pid [280910]
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessStarted'
[INFO] [scan_to_scan_filter_chain-5]: process started with pid [280911]
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessStarted'
[INFO] [joy_node-6]: process started with pid [280912]
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessStarted'
[INFO] [teleop_node-7]: process started with pid [280913]
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessStarted'
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7e8c34e4b440>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7e8c34bfb4a0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stdout.ProcessStdout object at 0x7e8c346d38c0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stdout.ProcessStdout object at 0x7e8c346d38c0>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x7e8c34778320>'
[micro_ros_agent-3] [1751612716.139407] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[micro_ros_agent-3] [1751612716.139568] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7e8c346d3350>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7e8c346d3890>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7e8c346d3260>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7e8c346d3860>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7e8c346d33b0>'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7e8c34c1ede0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7e8c34c1ede0>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x7e8c347da570>'
[scan_to_scan_filter_chain-5] [WARN] [1751612716.243772568] [rcl.logging_rosout]: Publisher already registered for node name: 'laser_filter'. If this is due to multiple nodes with the same name then all logs for the logger named 'laser_filter' will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7e8c34c1ebd0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7e8c34c1ebd0>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x7e8c346d2750>'
[teleop_node-7] [INFO] [1751612716.253714540] [TeleopTwistJoy]: Teleop enable button 4.
[teleop_node-7] [INFO] [1751612716.253794640] [TeleopTwistJoy]: Turbo on button 5.
[teleop_node-7] [INFO] [1751612716.253798608] [TeleopTwistJoy]: Linear axis x on 1 at scale 0.700000.
[teleop_node-7] [INFO] [1751612716.253807337] [TeleopTwistJoy]: Turbo for linear axis x is scale 1.500000.
[teleop_node-7] [INFO] [1751612716.253810258] [TeleopTwistJoy]: Linear axis y on 0 at scale 0.700000.
[teleop_node-7] [INFO] [1751612716.253812491] [TeleopTwistJoy]: Turbo for linear axis y is scale 1.500000.
[teleop_node-7] [INFO] [1751612716.253814895] [TeleopTwistJoy]: Angular axis yaw on 3 at scale 1.000000.
[teleop_node-7] [INFO] [1751612716.253817188] [TeleopTwistJoy]: Turbo for angular axis yaw is scale 1.500000.
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7e8c34ad34d0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7e8c34ad34d0>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x7e8c34911d00>'
[robot_state_publisher-1] [INFO] [1751612716.255382906] [robot_state_publisher]: Robot initialized
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7e8c34df10d0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7e8c34df10d0>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x7e8c347e2cf0>'
[ros2_control_node-2] [INFO] [1751612716.308472551] [controller_manager]: Using Steady (Monotonic) clock for triggering controller manager cycles.
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7e8c34e4b440>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7e8c34e4b440>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x7e8c347e2cf0>'
[ros2_control_node-2] [INFO] [1751612716.318981391] [controller_manager]: Subscribing to '/robot_description' topic for robot description.
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7e8c34e4b440>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7e8c34e4b440>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x7e8c347e2cf0>'
[ros2_control_node-2] [INFO] [1751612716.352708276] [controller_manager]: update rate is 100 Hz
[ros2_control_node-2] [INFO] [1751612716.352764085] [controller_manager]: Spawning controller_manager RT thread with scheduler priority: 50
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7e8c34e4b440>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7e8c34e4b440>' ✓ '<launch.event_handlers.on_process_io.OnProcessIO object at 0x7e8c347e2cf0>'
[... cut ...]
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x7e8c34db2120>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x7e8c34db2120>' ✓ '<launch.event_handlers.on_process_exit.OnProcessExit object at 0x7e8c347e2ff0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x7e8c34db2120>' ✓ '<launch.event_handlers.on_process_exit.OnProcessExit object at 0x7e8c347e3dd0>'
husarion@my-ROSbotXL:~/rosbot_ws_jazzy$ 
```

rafal-gorecki, 2025-07-15:
Hello [@Dhon_G](https://community.husarion.com/u/dhon_g),

The shoe issue you described is related to a bug in the lunch and lunch_ros packages. We don’t really have control over these packages, so we’ll just have to wait. I found more information:

[Using ``TimerAction`` with ``SetParameter`` from launch_ros causes crash by kbrameld · Pull Request #879 · ros2/launch · GitHub](https://github.com/ros2/launch/pull/879)
[Substitution normalization issue in TimerAction · Issue #894 · ros2/launch · GitHub](https://github.com/ros2/launch/issues/894)
For now, we recommend using the ROSbot [snap](https://snapcraft.io/rosbot) (latest changes can be found on jazzy/edge) or [Docker](https://github.com/husarion/rosbot_ros/blob/jazzy/docker/compose.rosbot_xl.yaml) approach.

Regards

rafal-gorecki, 2025-07-17:
Until the issue is resolved, you can download and build the package for your workspace (src) folder:

```bash
git clone -b backport-timer-reset https://github.com/rafal-gorecki/launch
```

I've also added a PR to the official project, but we'll have to wait for it to merge.

https://github.com/ros2/launch/pull/893

closed 2025-08-04 by rafal-gorecki